IDEC-ezWheel/swd_ros2_controllers #4: Build of source code fails

suguruhamada1003910, 2022-10-28:
When I execute "apt upgrade", build of source code failed. Do you have solution?

```
suguru@c-lati7280-076:~/ros2_ws$ colcon build
Starting >>> linetracer
Starting >>> swd_ros2_controllers
Starting >>> swd_ros2_starter_kit_bringup
Finished <<< swd_ros2_starter_kit_bringup [0.71s]        
--- stderr: swd_ros2_controllers                         
make[2]: *** No rule to make target '/opt/ros/galactic/lib/libfastrtps.so.2.3.4', needed by 'libswd_ros2_controllers__rosidl_typesupport_fastrtps_cpp.so'.  Stop.
make[1]: *** [CMakeFiles/Makefile2:475: CMakeFiles/swd_ros2_controllers__rosidl_typesupport_fastrtps_cpp.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
make: *** [Makefile:141: all] Error 2
---
Failed   <<< swd_ros2_controllers [0.79s, exited with code 2]
Aborted  <<< linetracer [1.14s]                     

Summary: 1 package finished [1.43s]
  1 package failed: swd_ros2_controllers
```

suguruhamada1003910, 2022-10-28:
I think the version of libfastrtps.so is updated.

```
suguru@c-lati7280-076:/opt/ros/galactic/lib$ ls | grep libfast
libfastcdr.so
libfastcdr.so.1
libfastcdr.so.1.0.20
libfastrtps.so
libfastrtps.so.2.3
libfastrtps.so.2.3.5

```

suguruhamada1003910, 2022-11-02:
I uninstalled fastrtps v2.3.5 and I build fastrtps v2.3.4. Then, this problem is not occurred.
https://github.com/eProsima/Fast-DDS/tree/v2.3.4

However, the fastrtps v2.3.5 is official package. So, I would like to know the way to modify source code of swd about the version of other libraries.

GMezWheel, 2022-11-13:
Hi @suguruhamada1003910,

I think that you need to remove build cache. I didn't reproduced your problem on my environment.

```
🐳 [ros2-galactic] swd_sk@nuvo: lib $ ls -al /opt/ros/galactic/lib/libfast*
lrwxrwxrwx 1 root root      15 Jan 24  2022 /opt/ros/galactic/lib/libfastcdr.so -> libfastcdr.so.1
lrwxrwxrwx 1 root root      20 Jan 24  2022 /opt/ros/galactic/lib/libfastcdr.so.1 -> libfastcdr.so.1.0.20
-rw-r--r-- 1 root root   80832 Jan 24  2022 /opt/ros/galactic/lib/libfastcdr.so.1.0.20
lrwxrwxrwx 1 root root      18 Aug 25 13:38 /opt/ros/galactic/lib/libfastrtps.so -> libfastrtps.so.2.3
lrwxrwxrwx 1 root root      20 Aug 25 13:38 /opt/ros/galactic/lib/libfastrtps.so.2.3 -> libfastrtps.so.2.3.5
-rw-r--r-- 1 root root 8894080 Aug 25 13:38 /opt/ros/galactic/lib/libfastrtps.so.2.3.5
```
```
cd ros2_ws
rm -rf build/ install/ log/
colcon build
Starting >>> swd_ros2_controllers
Finished <<< ez_way_bringup [0.74s]                                                                                                      
Finished <<< swd_ros2_controllers [17.1s]                       

Summary: 2 packages finished [17.3s]

```

suguruhamada1003910, 2022-11-14:
@GMezWheel 

I could build source code with your steps!! Thank you so much!!!

closed 2022-11-14 by suguruhamada1003910