IDEC-ezWheel/swd_ros2_controllers #5: ezw-swd-*.service will not start on clean install IPC

fig-shimooka, 2022-12-13:
Architecture: x86-64
OS: Ubuntu20.04
ROS:Galactic

I am initializing the OS and setting it up from scratch.
Both ezw-swd-right and ezw-swd-left fail to boot.
. /remote also fails to run.
The only services that are running are ezw-stack.service and ezw-dbus-user-session.
The can0 is up, but the candump results show nothing.

```
● ezw-swd-right.service - EZW SWD RIGHT SERVICE
     Loaded: loaded (/etc/systemd/system/ezw-swd-right.service; enabled; vendor preset: enabled)
     Active: activating (auto-restart) (Result: exit-code) since Tue 2022-12-13 08:53:04 UTC; 107ms ago
    Process: 15383 ExecStart=/opt/ezw/sbin/sce-swd-right.sh start (code=exited, status=1/FAILURE)

```

```
fig@x86:/opt/ezw/usr/sbin$ ./remote.py swd_left
SYSTEMCTL_dbus.id detected
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/dbus/bus.py", line 177, in activate_name_owner
    return self.get_name_owner(bus_name)
  File "/usr/lib/python3/dist-packages/dbus/bus.py", line 361, in get_name_owner
    return self.call_blocking(BUS_DAEMON_NAME, BUS_DAEMON_PATH,
  File "/usr/lib/python3/dist-packages/dbus/connection.py", line 652, in call_blocking
    reply_message = self.send_message_with_reply_and_block(
dbus.exceptions.DBusException: org.freedesktop.DBus.Error.NameHasNoOwner: Could not get owner of name 'commonapi.SmcService.NMT.v1_0_commonapi.ezw.smcservice.swd_left': no such name

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "./remote.py", line 549, in <module>
    main(sys.argv[1:])
  File "./remote.py", line 531, in main
    nmt_client = NMTDBusClient(instance_id)
  File "/opt/ezw/usr/lib/smcdbusclient/nmt.py", line 49, in __init__
    self.dbus_object = self.bus.get_object(
  File "/usr/lib/python3/dist-packages/dbus/bus.py", line 241, in get_object
    return self.ProxyObjectClass(self, bus_name, object_path,
  File "/usr/lib/python3/dist-packages/dbus/proxies.py", line 250, in __init__
    self._named_service = conn.activate_name_owner(bus_name)
  File "/usr/lib/python3/dist-packages/dbus/bus.py", line 182, in activate_name_owner
    self.start_service_by_name(bus_name)
  File "/usr/lib/python3/dist-packages/dbus/bus.py", line 277, in start_service_by_name
    return (True, self.call_blocking(BUS_DAEMON_NAME, BUS_DAEMON_PATH,
  File "/usr/lib/python3/dist-packages/dbus/connection.py", line 652, in call_blocking
    reply_message = self.send_message_with_reply_and_block(
dbus.exceptions.DBusException: org.freedesktop.DBus.Error.ServiceUnknown: The name commonapi.SmcService.NMT.v1_0_commonapi.ezw.smcservice.swd_left was not provided by any .service files
fig@x86:/opt/ezw/usr/sbin$ ./remote.py swd_right
SYSTEMCTL_dbus.id detected
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/dbus/bus.py", line 177, in activate_name_owner
    return self.get_name_owner(bus_name)
  File "/usr/lib/python3/dist-packages/dbus/bus.py", line 361, in get_name_owner
    return self.call_blocking(BUS_DAEMON_NAME, BUS_DAEMON_PATH,
  File "/usr/lib/python3/dist-packages/dbus/connection.py", line 652, in call_blocking
    reply_message = self.send_message_with_reply_and_block(
dbus.exceptions.DBusException: org.freedesktop.DBus.Error.NameHasNoOwner: Could not get owner of name 'commonapi.SmcService.NMT.v1_0_commonapi.ezw.smcservice.swd_right': no such name

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "./remote.py", line 549, in <module>
    main(sys.argv[1:])
  File "./remote.py", line 531, in main
    nmt_client = NMTDBusClient(instance_id)
  File "/opt/ezw/usr/lib/smcdbusclient/nmt.py", line 49, in __init__
    self.dbus_object = self.bus.get_object(
  File "/usr/lib/python3/dist-packages/dbus/bus.py", line 241, in get_object
    return self.ProxyObjectClass(self, bus_name, object_path,
  File "/usr/lib/python3/dist-packages/dbus/proxies.py", line 250, in __init__
    self._named_service = conn.activate_name_owner(bus_name)
  File "/usr/lib/python3/dist-packages/dbus/bus.py", line 182, in activate_name_owner
    self.start_service_by_name(bus_name)
  File "/usr/lib/python3/dist-packages/dbus/bus.py", line 277, in start_service_by_name
    return (True, self.call_blocking(BUS_DAEMON_NAME, BUS_DAEMON_PATH,
  File "/usr/lib/python3/dist-packages/dbus/connection.py", line 652, in call_blocking
    reply_message = self.send_message_with_reply_and_block(
dbus.exceptions.DBusException: org.freedesktop.DBus.Error.ServiceUnknown: The name commonapi.SmcService.NMT.v1_0_commonapi.ezw.smcservice.swd_right was not provided by any .service files
```

ez-Support, 2022-12-13:
You might have to export your dbus session 1st in your terminal

`export $(cat /tmp/SYSTEMCTL_dbus.id)  `

fig-shimooka, 2022-12-15:
export $(cat /tmp/SYSTEMCTL_dbus.id) 
I ran this and it solved the problem.
Thanks.