IDEC-ezWheel/swd_ros2_controllers #6: Committioning when using two SE2Ls

fig-shimooka, 2022-12-15:
This configuration constitutes the SWD.
I want to enable SDI, SLS in forward and backward.
The default is that SDI, SLS is enabled only for forward.
I would like to enable them in the back as well.
How do I configure the Commitioning settings?

[image]
[image]
[image]

ez-Support, 2022-12-16:
Please find in the enclosed zip folder, the modification for :
`/SafetyHub12pts/swd_left_4_commissioning.py` and 
`/SafetyHub12pts/swd_right_5_commissioning.py`

[swd-starter-kit-config-2xSE2l.zip](https://github.com/ezWheelSAS/swd_ros2_controllers/files/10246142/swd-starter-kit-config-2xSE2l.zip)

Please check the wiring on the SafeIO connector of the SWD, and in the SafeIn field of the `remote.py` diagnostic.
You should be able to see the OSSDs switching from false to true on the right wheel, when the scanner commuts one field.

[image]

[image]

ez-Support, 2022-12-19:
Are the Safety fonctions activated, i.e. SDI or SLS ?

If not, can you 'start all nodes', with the remote.py, keyword 'm'. And then press 'o' to make them operational.

fig-shimooka, 2022-12-20:
swd left
[image]

swd right
[image]

fig-shimooka, 2022-12-20:
Since the safety topic has not been PUBLISHED, it is not possible to determine if the SWD is in SDI or STO status.
Also, both OSSD LEDs on SE2L are green (no obstructions)

fig-shimooka, 2022-12-21:
What's this error ... ? 

sh: 1: cansend: not found

fig-shimooka, 2022-12-21:
I didn't install can-utils.

fig-shimooka, 2022-12-21:
The result is the same.

left
[image]

right
[image]

ez-Support, 2022-12-21:
Please send the whole diagnostic after having pressed '`m`', then '`o`'.

fig-shimooka, 2022-12-22:
The results are put up.
This is all that is shown

> The result is the same.
> 
> left [image]
> 
> right [image]

fig-shimooka, 2023-01-06:
Is it possible to get a sample of Commissioning when SDI is enabled in a configuration with two SE2Ls connected?

fig-shimooka, 2023-01-08:
This is the result of pressing in the order m->o->a.
Also, SAFEIN_1 seemed to work correctly for both RIGHT and LEFT.
When 1/2 of OSSD responded, SLS of SAFEIN_1 was False, and when 3/4 responded, SDI was False.
However, when I actually PUBLISH cmd_vel, the motor does not rotate. (even when SDI and SLS are not reacting)

[image]
[image]

fig-shimooka, 2023-01-10:
[image]

fig-shimooka, 2023-01-10:
. /remote.py After executing swd_right, c(log clear) was executed and "e" was executed.
SRDO_SCT_timeout is occurring periodically.

[image]

fig-shimooka, 2023-01-10:
this is can log.

[can_log.txt](https://github.com/ezWheelSAS/swd_ros2_controllers/files/10380997/can_log.txt)

ez-Support, 2023-01-10:
Can you retry to do the commissioning on swd_left ?
It seems that a CANopen dico value is corrupted:
[image]

fig-shimooka, 2023-01-10:
After running Commission several times, the error disappeared.
The motor is now running.
But SDI does not work as intended.

When moving forward, the motor stops when there is an obstacle in front of the SE2L (=SDI). And backward movement is possible.

But when going backward, when there is an obstacle in front of the two SE2Ls, the motor will neither stop nor slow down.
あ
And when there is an obstacle in the back SE2L, it can back up.
However, it is able to move forward.

In other words, when there is an obstacle in front or behind, in either direction, forward SDI is enabled.

[image]

[image]

fig-shimooka, 2023-01-10:
commissioning script
[Uploading com_20230110.zip…]()

fig-shimooka, 2023-01-10:
commissoning script
[com_20230110.zip](https://github.com/ezWheelSAS/swd_ros2_controllers/files/10382263/com_20230110.zip)

ez-Support, 2023-01-10:
I think I've been mistaken, and I have inverted SDIN and SDIP.
Can you please retrye with the commissioning 12pts below:

[com_20230110_bis.zip](https://github.com/ezWheelSAS/swd_ros2_controllers/files/10383662/com_20230110_bis.zip)

fig-shimooka, 2023-01-12:
[image]

fig-shimooka, 2023-01-12:
I rewrote the parameters about 5 or 6 times and it worked.
Why did the parameter error occur again and again?

Also, with the current hardware configuration, SAFEIN2 and 3 are SLS and SAFEIN4 and 5 are SDI (that's how the electrical wiring is).

Is this supposed to be the other way around?

ez-Support, 2023-01-12:
Yes, indeed it should be inverted.

First SAFEIN pair = STO
2nd SAFEIN pair = SDI
3rd SAFEIN pair = SLS

ez-Support, 2023-01-12:
For the error parameter, is the ros-bringup service stopped ? You can try stopping all ros services during commissioning.

fig-shimooka, 2023-01-12:
ros-bringup is not used. Instead, launch.py is run.
[image]

I stop swd-ros2-controller when commisioning.
However, the parameter is an error.

ez-Support, 2023-01-13:
I suggest anyway to stop all ros applications running during commissioning. 
Is the mapping correct regardind the SDI and SLS functions now ?

fig-shimooka, 2023-01-15:
Now you are right.
This issue is closed.

ez-Support, 2023-01-16:
Did the error parameter stopped after having stopped ros applications running ?

fig-shimooka, 2023-01-27:
When all ROS nodes were stopped, the parameters were successfully changed.

closed 2023-03-10 by GMezWheel