IDEC-ezWheel/swd_ros2_controllers #11: Deceleration when the motor is in SDI

fig-shimooka, 2023-03-29:
Is it possible to change the deceleration when the motor goes into SDI?
Is the setting of vl_dec_delta_speed reflected?

https://github.com/ezWheelSAS/swd-starter-kit-config/blob/main/commissioning/SafetyHub12pts/swd_left_4_commissioning.py#L98

ez-Support, 2023-03-29:
Actually, SDI violation activates a STO, and the STO activates the SBC in the actual configuration (cf. objet CANopen 0x6643:00).

In the new firmware release 2.0.0, you will be able to choose the type of SBC you want to activate (and also no SBC, i.e. no brake).

So the answer is yes, but not yet.. The firmware release 2.0.0 will be released in a few weeks.

fig-shimooka, 2023-03-29:
thank you.
this isshue is close.

closed 2023-08-22 by GMezWheel