IDEC-ezWheel/swd_ros2_controllers #13: Can't install on ROS2 Foxy arm64 machines

HirokiShikazaki, 2023-05-25:
Hello.
I am trying to install swd_ros2_controllers on Raspberry Pi 3 model B arm64 machines but I get an error.
repository and GPG key have already been added.
I'm using ROS2Foxy.

The command I ran was
"sudo apt update && sudo apt install swd-services ros-foxy-swd-ros2-controllers"

The error message was 
"E: Unable to locate package ros-foxy-swd-ros2-controllers"

I look forward to your feedback
Best Regards,

ez-Support, 2023-05-25:
I think you missed to add the URL localistation of ez-Wheel debian packages:

```
sudo echo "deb http://packages.ez-wheel.com:8081/apt-repo focal main" >> /etc/apt/sources.list
sudo wget -qO - http://packages.ez-wheel.com:8081/archive.key | apt-key add -
sudo apt-get update
sudo apt install swd-services
sudo apt install ros-foxy-swd-ros2-controllers
```

GMezWheel, 2023-05-25:
Hi @HirokiShikazaki,

I confirm that the ROS2 foxy package for ARM64 doesn't exist.
Can you build it by yourself ?

HirokiShikazaki, 2023-05-26:
Hi @GMezWheel 

Thank you for response.
I could build the ROS2 foxy package for ARM64 by Raspberry Pi 3.
I would appreciate it if you could modify the apt install method.

Best Regards.

GMezWheel, 2023-05-26:
Yes, of course. I'm adding it into the backlog.

HirokiShikazaki, 2023-10-05:
Hi
Could you please modify the ROS2 foxy package for ARM64 mentioned in a other day comment so that the apt install method is possible?

GMezWheel, 2023-12-06:
Hi @HirokiShikazaki,
The ROS2 foxy package for ARM64 is now available : **ros-foxy-swd-ros2-controllers_2.0.0-0focal_arm64.deb**.
It depends of the new swd-services package : **swd-services_2.0.0_arm64.deb**

shamada-idec, 2023-12-20:
@GMezWheel 

I could not install ros-foxy-swd-ros2-controllers and swd-services v2.0.0 in raspberry pi 4.
I checked them with 'dpkg -l | grep swd'. There is not "ros-foxy-swd-ros2-controllers" package and the latest version of swd-services is 0.2.7.

I already could installed swd-services v0.2.7. So, I think the my setting of sources.list is correct.

Could you check it again?

shamada-idec, 2023-12-20:
@GMezWheel 

I don't know why, but when I tried it again, I was able to install them well.
Please ignore above comments.

GMezWheel, 2023-12-20:
@shamada-idec,
As a reminer, in order to see new packages, you have to update your package list first: 
`sudo apt update`

closed 2023-12-20 by ez-Support