IDEC-ezWheel/swd_ros2_controllers #16: Package server unreachable

TheoBiardeau, 2023-09-08:
Hello,
I followed your github instructions to install your package for ROS2, but I get the following error message after the `sudo apt update`:
`Failed to fetch http://packages.ez-wheel.com:8081/apt-repo/dists/focal/InRelease  Could not connect to packages.ez-wheel.com:8081 (51.83.77.229), connection timed out
`
It doesn't matter whether the installation is done in a docker or in a host, the error is the same.

What is the procedure to resolve this problem?

Best,
Théo

ez-Support, 2023-09-08:
Hello, 

You might need to add the ez-Wheel repository to your Apt sources configuration file, to be able to download our packages:

`echo "deb http://packages.ez-wheel.com:8081/apt-repo focal main" | sudo tee -a /etc/apt/sources.list`

TheoBiardeau, 2023-09-08:
Hello,

The `deb http://packages.ez-wheel.com:8081/apt-repo focal main` is present in my sources.list.
When I type the following command: 
`wget -O - http://packages.ez-wheel.com:8081/archive.key
--2023-09-08 13:36:19--  http://packages.ez-wheel.com:8081/archive.key
Resolving packages.ez-wheel.com (packages.ez-wheel.com)... 51.83.77.229
Connecting to packages.ez-wheel.com (packages.ez-wheel.com)|51.83.77.229|:8081... 
`
I cannot reach the server, even after a few minutes.

ez-Support, 2023-09-08:
Try :
`http://packages.ez-wheel.com:8081/apt-repo/pool/`

ez-Support, 2023-09-08:
You can try to connect with a smartphone WiFi hotspot, as sometimes companies firewall does not grant the access to our packages URL.

TheoBiardeau, 2023-09-08:
Problem solved. This was due to university network restrictions.
Thanks for your help.
Théo

closed 2023-09-08 by TheoBiardeau

ez-Support, 2023-09-08:
You can ask them to add the URL, to remove firewall restrictions 😉