IDEC-ezWheel/swd_ros2_controllers #18: Cannot get Humble package by apt command

shamada-idec, 2024-01-15:
I tried to install humble package according to README but I cannot install it.

```
Humble
sudo apt update && sudo apt install swd-services ros-humble-swd-ros2-controllers
```

Error is below.
```
E: Unable to locate package ros-humble-swd-ros2-controllers
```

GMezWheel, 2024-01-15:
Hi @shamada-idec ,

Indeed, there is a mistake into the README.txt, in order to install humble/iron ROS2 packages, you need to add ez-Wheel **jamy** repository (instead of **focal**):
`echo "deb http://packages.ez-wheel.com:8081/apt-repo jamy main" | sudo tee -a /etc/apt/sources.list`

NOTA : be sure to remove the ez-Wheel focal apt-repo line from /etc/apt/sources.list

shamada-idec, 2024-01-15:
> Hi @shamada-idec ,
> 
> Indeed, there is a mistake into the README.txt, in order to install humble/iron ROS2 packages, you need to add ez-Wheel **jammy** repository (instead of **focal**): `echo "deb http://packages.ez-wheel.com:8081/apt-repo jammy main" | sudo tee -a /etc/apt/sources.list`
> 
> NOTA : be sure to remove the ez-Wheel focal apt-repo line from /etc/apt/sources.list

Thank you for reply. I will try it.

shamada-idec, 2024-01-16:
I could install the package!!

closed 2024-01-16 by ez-Support