IDEC-ezWheel/swd_ros2_controllers #20: Velocity Limitation

elkuno213, 2024-08-26:
Hello,

I am currently trying to increase the velocity, which is currently limited to 1 m/s. I have attempted to adjust the `motor_max_speed_rpm` and `motor_max_safety_limited_speed_{1,2}_rpm` parameters to their maximum of 2000 RPM (which corresponds to approximately 3 m/s). However, the velocity remains limited to 1 m/s.

To further investigate, I attempted to hardcode the 2000 RPM value directly in the controller API DBusClient::setTargetVelocity() as shown below. Unfortunately, the result remains unchanged, with the velocity still limited to 1 m/s.

Is there another method or setting to increase the velocity beyond this limit?

**Additional Note**: When the wheel is limited to 1 m/s, I also notice a persistent 'clac clac' noise. This noise seems to occur consistently under this limitation.

```cpp
/**
 * @brief Indicates the required velocity of the system. It will be multiplied by
 * the velocity dimension factor and the velocity set-point factor. The value should be given
 * in user-defined velocity units or in revolutions per minute (r/min), if the
 * velocity dimension factor and the velocity set-point factor are not implemented or have
 * a value of 1. Positive values indicate forward direction, and negative values indicate reverse direction.
 *
 * @param pTargetVelocity The required velocity (range: -2000 to 2000 r/min).
 * @return ERROR_NONE if no error occurred.
 */
virtual ezw_error_t setTargetVelocity(const int16_t &pTargetVelocity) override;
```

Thank you for your help.

ez-Support, 2024-08-29:
After having changed **motor_max_speed_rpm**, did you recompile the ROS node and relaunched the node afterwards ?

elkuno213, 2024-08-30:
Yes, we followed all the necessary steps. The issue actually stemmed from the commissioning configuration, where the `sls_vl_limit` parameter was hardcoded to 680 RPM for both the [left](https://github.com/ezWheelSAS/swd-starter-kit-config/blob/4d1808ddef47d051eae8d9a784e209da898a0a37/commissioning/swd_left_4_commissioning.py#L98) and [right](https://github.com/ezWheelSAS/swd-starter-kit-config/blob/4d1808ddef47d051eae8d9a784e209da898a0a37/commissioning/swd_right_5_commissioning.py#L98) wheels during commissioning. After modifying these values and re-commissioning both wheels, the problem was resolved.

It's important to note that the parameters `motor_max_speed_rpm` and `motor_max_safety_limited_speed_{1,2}_rpm` are only used for limiting commands at the ROS level. Therefore, it would be beneficial to document the commissioning configuration and clarify how it differs from the ROS node parameters.

GMezWheel, 2024-09-06:
Hi @elkuno213 ,

I'm pleased to see that the problem has been solved.

Indeed, the ROS parameters must be consistent with those used for commissioning. 
Thank you for this comment, I will specify it in the documentation.

Concerning the "clack, clack" noise, it is due to SLS speed limit violation :
=> (velocity > SLS limit) => (SLS violation) => (velocity = 0) **clack** => (velocity < SLS limit) => (velocity > SLS limit) => (SLS violation) => (velocity = 0) **clack** => ...

closed 2024-09-06 by GMezWheel