IDEC-ezWheel/swd_ros2_controllers #21: Acceleration behaviour

jmartherose-aldebaran, 2024-09-02:
Hello, 

We've been using the *swd_ros2_controller* to run our robot equipped with two (left and right) ezWheels with ROS2, and we're observing a behavior where it appears that the time to reach a given speed is constant, no matter the final speed we're trying to achieve, rather than being dependent on time.

For example:
* if we're trying to go at 0.5m/second, the robot will take ~7 second to reach the target speed
* if we're trying to go at 1.5m/second, the robot will take ~7 second to reach the target speed

Is it possible to change it so that instead, we'd have something like:
* if we're trying to go at 0.5m/second, the robot will take ~3.5 seconds to reach the target speed
* if we're trying to go at 1m/second, the robot will take ~7 seconds to reach the target speed

(Which I IIRC is the default behavior expected by ROS navigation-related plugins)

The parameters related to the velocity mode / ramp behavior have been left to default because we're unsure on how to modify them.

Thank you for your help.

ez-Support, 2024-09-06:
Hello, 

According to your SWD gear ratio .4, 
You can test to decrease the acceleration ramps as follow, so the robot will accelerate **more slowly:**

    vl_acc_delta_speed = 100  # 1500
    vl_dec_delta_speed = 500  # 1500

For both wheels in the commissioning:
 https://github.com/ezWheelSAS/swd-starter-kit-config/blob/main/commissioning/swd_left_4_commissioning.py#L95 
https://github.com/ezWheelSAS/swd-starter-kit-config/blob/main/commissioning/swd_right_5_commissioning.py#L95

jmartherose-aldebaran, 2024-09-13:
Thank you for the answer, I will try these parameters ASAP and report back here with the results.