IDEC-ezWheel/swd_ros2_controllers #22: Activating SBC with remote.py doesn't persist after starting swd_diff_drive_controller.launch.py

louiskenzo, 2024-09-10:
Hello,

Trying to configure SBC on a SWD 125 wheel, by running `remote.py` and pressing the `t` key, setting SBC to 1.

```
SYSTEMCTL_dbus.id detected
DBUS Instance        : swd_left ..................................................................... [OK]
Core Version         : 2.0.1-daa5833 ................................................................ [OK]
Core connected       : True ......................................................................... [OK]
Core PDO initialized : True ......................................................................... [OK]
Core CAN device      : can0 ......................................................................... [OK]
Node ID              : 0x4 .......................................................................... [OK]
RT activated         : True ......................................................................... [OK]
Vendor-Id            : 0x515 ........................................................................ [OK]
Product code         : 0x11 ......................................................................... [OK]
Revision number      : 2.0 .......................................................................... [OK]
Serial number        : 24179SC012A1 ................................................................. [OK]
Product Ref          : ezSWD125IM.4B/C .............................................................. [OK]
SW Ref               : FW_SWD_CORE .................................................................. [OK]
SW Version           : 2.0.2 ........................................................................ [OK]
HW Id                : 777H ......................................................................... [OK]
Calibrated           : True ......................................................................... [OK]
SRDO validity        : True ......................................................................... [OK]
NMT state on error   : 1 ............................................................................ [OK]
NMT State            : OPER  ........................................................................ [OK]
PDS State            : OPERATION_ENABLED  ........................................................... [OK]
NbError              : 18 ........................................................................... [OK]
LastError            : 800E : EMERGENCY ............................................................. [OK]
SystemError          : EZW_PROTECT_NONE EZW_PROTECT_NONE  ........................................... [OK]
SSW CAN1             : False, False, False, False, False, False, False, False  ...................... [OK]
SSW CAN2             : False, False, False, False, False, False, False, False  ...................... [OK]
SSW CAN3             : False, False, False, False, False, False, False, False  ...................... [OK]
SSW CAN4             : False, False, False, False, False, False, False, False  ...................... [OK]
SSW CAN5             : False, False, False, False, False, False, False, False  ...................... [OK]
SSW CAN6             : False, False, False, False, False, False, False, False  ...................... [OK]
SSW CAN7             : False, False, False, False, False, False, False, False  ...................... [OK]
SSW CAN8             : False, False, False, False, False, False, False, False  ...................... [OK]
SSW SAFEOUT          : False, False, False, False, False, False, False, False  ...................... [OK]
SCW SAFEIN_1         : True, True, True, True, False, False, True, True  ............................ [OK]
SCW PERMANENT CAN1   : NONE, NONE, NONE, NONE, NONE, NONE, NONE, NONE  .............................. [OK]
SCW PERMANENT CAN2   : NONE, NONE, NONE, NONE, NONE, NONE, NONE, NONE  .............................. [OK]
SafetyFunctionOutput : [SLS_1, ERROR_ACK] ........................................................... [OK]
SystemErrorOutput    : [] ........................................................................... [OK]
VelocityModeSW       : internal_limit_active: False ................................................. [OK]
VelocityModeCW       : enable_ramp: False, unlock_ramp: False, reference_ramp: False, halt: False ... [OK]
Polarity             : velocity_polarity: True, position_polarity: True ............................. [OK]
ExternalBrakePresent : True ......................................................................... [OK]
STO                  : restart_acknowledge_behavior: False, activate_sbc: 0x66600100 ................ [OK]
TargetVelocity       : 0 ............................................................................ [OK]
VelocityDemand       : 0 ............................................................................ [OK]
VelocityActualValue  : 0 ............................................................................ [OK]
PositionValue        : 203817 ....................................................................... [OK]
OdometryValue        : 666989 ....................................................................... [OK]
<COM> Connect NodeId = i Disconnect = d Restore param = j Change current NodeId = g
<NMT> RESET_NODE = r Start all = m
<PDS> Go to OPERATION_ENABLED = o
<VL mode> target velocity (500) = +/- positive order = 8 negative order = 2 stop = *
<Param> STO_ACK off/on = f/n Change brake_presence to False = b
<Mode> local = w remote = z
<Diagnostic> a
<STO> change STO activate_SBC = t
<EMCY> get full EMCY list = e, clear EMCY list = c
<SW> s
<quit> 0
current config for STO active SBC is 1717567744
Enter the number of the SBC you want set to STO activate SBC (0, 1, 2 or 3) [1]: 1
Do you want to save this parameters ? (This parameter has to be correlated with the brake present) [y/N]: y
<COM> Connect NodeId = i Disconnect = d Restore param = j Change current NodeId = g
<NMT> RESET_NODE = r Start all = m
<PDS> Go to OPERATION_ENABLED = o
[... 110 lines cut ...]
SystemErrorOutput    : [] ........................................................................... [OK]
VelocityModeSW       : internal_limit_active: False ................................................. [OK]
VelocityModeCW       : enable_ramp: False, unlock_ramp: False, reference_ramp: False, halt: False ... [OK]
Polarity             : velocity_polarity: True, position_polarity: True ............................. [OK]
ExternalBrakePresent : True ......................................................................... [OK]
STO                  : restart_acknowledge_behavior: False, activate_sbc: 0x66600100 ................ [OK]
TargetVelocity       : 0 ............................................................................ [OK]
VelocityDemand       : 0 ............................................................................ [OK]
VelocityActualValue  : 0 ............................................................................ [OK]
PositionValue        : -69 .......................................................................... [OK]
OdometryValue        : -225 ......................................................................... [OK]
<COM> Connect NodeId = i Disconnect = d Restore param = j Change current NodeId = g
<NMT> RESET_NODE = r Start all = m
<PDS> Go to OPERATION_ENABLED = o
<VL mode> target velocity (500) = +/- positive order = 8 negative order = 2 stop = *
<Param> STO_ACK off/on = f/n Change brake_presence to False = b
<Mode> local = w remote = z
<Diagnostic> a
<STO> change STO activate_SBC = t
<EMCY> get full EMCY list = e, clear EMCY list = c
<SW> s
<quit> 0
```

What is the recommended way of configuring SBC for use with the ROS2 driver?

ez-Support, 2024-09-11:
Hello, 

When you run the diff_drive_controller, PDS State is switched to OPERATION_ENABLED. 
As soon as you have this state, the SBC (brake) is released, so it is normal no seeing it.

You can check pressing an e-stop, you will see SBC_1, in the SafetyFunctionOutput

ez-Support, 2024-09-12:
It can be usefull to change the PDS State going downhill on a slope, so that the robot can be braked.
This will require to modify the diff_drive_controller, programming the PDS State to switch to another state, when cmd_vel is null for instance

closed 2024-10-25 by louiskenzo